m5stack/M5-Schematic #1: GROVE connector GPIO12 -> 4 ?

sige1, 2018-12-09:
MODEL:M5CAMERA001
It was confirmed operation with MicroPython (psram).

GPIO 4 = Low : LED Light On
GPIO 4 = High : LED Light Off

thank you.

watson8544, 2018-12-12:
Thanks sige1. I am going to update it. The new version of M5CAMERA, pinmap of BME280 is scl -> 23, sda -> 22.  Recently, we'll update a new code for M5CAMERA according to espressif camera reposity.

https://github.com/espressif/esp32-camera

watson8544, 2018-12-17:
> MODEL:M5CAMERA001
> It was confirmed operation with MicroPython (psram).
> 
> Manual connect:
> GROVE connector --> BME280
> 
> import machine
> i2c = machine.I2C(scl=machine.Pin(4), sda=machine.Pin(13))
> i2c.scan()
> 
> 118 <- get slave address 0x76(BME280)
> 
> thank you.

Hello,
Here's the new firmware for M5Camera.

watson8544, 2018-12-17:
> MODEL:M5CAMERA001
> It was confirmed operation with MicroPython (psram).
> 
> Manual connect:
> GROVE connector --> BME280
> 
> import machine
> i2c = machine.I2C(scl=machine.Pin(4), sda=machine.Pin(13))
> i2c.scan()
> 
> 118 <- get slave address 0x76(BME280)
> 
> thank you.

Hello,
Here's the new firmware for M5Camera.
https://github.com/m5stack/m5stack-cam-psram

closed 2019-03-12 by sio-funmatsu