m5stack/M5-Schematic #2: M5Camera schematic mistake

nnn112358, 2018-12-16:
M5Camera schematic "Pixel Data Bit 4" and "Pixel Data Bit 7" is IO39.
Probabry, this is mistake.
https://github.com/m5stack/M5-Schematic/blob/master/Units/m5camera/hardware_diff_with_ESP32CAM_M5Camera.md

watson8544, 2018-12-17:
> M5Camera schematic "Pixel Data Bit 4" and "Pixel Data Bit 7" is IO39.
> Probabry, this is mistake.
> https://github.com/m5stack/M5-Schematic/blob/master/Units/m5camera/hardware_diff_with_ESP32CAM_M5Camera.md

Thanks gay!

Pixel Data Bit 4 -> IO39  
Pixel Data Bit 7-> IO19       

I have modified it.

sio-funmatsu, 2018-12-19:
@watson8544 no gay...guy lol

andywang2039, 2019-03-11:
Hi Watson&Sio-funmatsu,

I bought 2 pcs M5cameraX/F from your sale channel and will use it for second develop in IoT domain, i saw the M5cameraX/F  schematic is fragmentary, and it is not helpful us to deep use it, any possible you can share the complete M5cameraX/F  schematic to me? thanks your help in advance!
Andy

sio-funmatsu, 2019-03-12:
@andywang2039 

Sorry, We can't provide complete schematic document now.
What do you want to know? Please tell me in concrete terms.

andywang2039, 2019-03-12:
@sio-funmatsu thanks your reply, i have two question about your schematic.
1. schematic showed 2 pc LED, one is red and another is blue. but actual sample only has 1 pc red LED.
2. schematic showed is USB_Micro, but actual sample mounted is type-c
if possible, please help provide these two segment clear schematic for me study.

thanks

sio-funmatsu, 2019-03-12:
@andywang2039 

I answer your questions.
Both questions are our mistakes.

1. There is only one LED on M5Camera F / X.
2. Micro_USB is wrong → USB Type-C is correct.

so sorry...and thank you for your advice.

andywang2039, 2019-03-13:
@sio-funmatsu 
Thanks your answer.
I contacted your colleague in parallel who is in China  , but she tell me the schematic should be depend on your side and unavailable now. 
More questions ;
1. Our end application request USB type is Micro_USB,if we insist use Micro_USB, then any preformation impact for M5cameraX/F?  
2.I found the USB cable is very short, what original intention is? Long will impact data transmission and reduce power?
3.I heard OV2640 is EOL, how to ensure M5cameraX/F continue product life?

Andy

nnn112358, 2020-08-17:
The latest document has been updated, so close this.

https://github.com/m5stack/M5-Schematic/blob/master/Units/m5camera/CameraComparison_en.md

closed 2020-08-17 by nnn112358