m5stack/M5-Schematic #6: Clarity regarding MPU wired

jithin917, 2019-03-25:
In the file /M5-Schematic-master/Core/m5stick/EspCore.pdf  "MPU9250" is wired without connecting the interrupt pin. So the DMP(digital motion processor) in this unit cannot be used. Senodia SH200Q datasheet is also provided in the repository. Both works in operating voltage of 2.6v to 3.6v but later doesnot have a DMP inside. So which mpu is wired ? does it have a dmp ? where to the interrupt pin of mpu 9250 is connected in esp32 pin?. Thank you

watson8544, 2019-04-18:
sorry. the interrupt pin of mpu 9250 of stick was not connected to esp gpio.

jithin917, 2019-04-18:
Is there any way I can use the dmp ? Any sample code ?Thank you sir..

watson8544, 2019-04-19:
Are you familiar with Embedded programming? I found some code in Github that may help you. You can have a try.

watson8544, 2019-04-19:
https://github.com/WHeiu/Uav-control-module/tree/master/%E5%BC%80%E5%8F%91/src/Stm32_New_Program/Src/MPU9250

Sorry, I have not ever used mpu9250 with DMP.

jithin917, 2019-04-19:
Thank you sir. I will try to understand the code. Thank you very much.

watson8544, 2019-04-23:
You are welcom

watson8544, 2019-04-23:
Not at all

closed 2019-04-23 by watson8544