m5stack/M5-Schematic #7: M5Stick C power management schematic mistake (AXP192 vs IP5306)

arzaman, 2019-05-09:
seems there is an incongruence in the schematic for power management of M5stick C

https://github.com/m5stack/M5-Schematic/blob/master/Core/m5stick/Power.pdf

the schema reports IP5306 IC while the API documentation reports the AXP192 IC
https://docs.m5stack.com/#/en/api/axp192_m5stickc

can you update the schematic with the right version ??

thks

hattesen, 2019-10-18:
How come nobody have even bothered to comment on this issue after 6 months?
It is very difficult for M5 customers to work in an information vacuum, where fundamental information about power control IC is misleading and/or not made available to the customers/users.

Very disappointing!

f4grx, 2019-10-18:
M5 stack products are cool but have errors and are very badly documented. Still no schematic for M5stack FIRE CORE yet.

So it's unreliable.
I'm a bit disappointed too.