m5stack/M5-Schematic #8: Core Schematic Mistakes

zshivers, 2019-05-29:
There are a number of schematic issues:
1. Passive components around the IP5306 don't have reference designators. The output capacitors don't have values.
1. U10 does not have any pins for IIC
1. Section with U4 is labelled "Speaker", but it relates to EA3036 which is a power management chip.
1. Link for EA3036 datasheet is to an intranet address (192.13.102.98)

zshivers, 2019-07-18:
@sio-funmatsu @watson8544

f4grx, 2019-10-13:
Also schematic for Fire Core is not available. No info about PSRAM pinout with a potentially problematic with BUS C connector (GPIO16,17 used for Groove UART, seems also used for PSRAM, so it's a conflict?)

Please publish the schematic for M5Stack Fire Core. Please again.

Gitshaoxiang, 2019-10-14:
The GPIO 16 / 17 in Fire is connected to the PSRAM by default, so when you connect or stack other function modules, you need to avoid conflicts with these two pins to prevent the device from working properly and causing instability.

f4grx, 2019-10-18:
This is not an answer.
Please tell me

If I use PSRAM, GROVE PORT C  is unusable

Yes or no ?

Gitshaoxiang, 2019-10-18:
YES, You can only use one of them（PSRAM or PORT C）

closed 2019-10-21 by Gitshaoxiang