m5stack/M5-Schematic #9: M5StickC schematic ?

Oliv4945, 2019-06-24:
Hi,

It would be nice if you could also add M5StikckC schematics :)

Thank you

mrb0y, 2019-09-02:
I second that motion. (Parliamentary Procedure).

barabanus, 2019-11-01:
It would be cool to see how your connected MPU6886. Given the code it seems like you haven't connected its interrupt pin.

Did you consider how to sample MPU6886 data with a constant rate without interrupt connected?

diogoviannaaraujo, 2019-12-07:
Without schematic we cannot find out about interrupts, peripheral power management etc, please, it would take some simple but long reverse engineering to find most of these stuff out. Save us that time, release the schematics.

Tinyu-Zhao, 2022-08-22:
https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/schematic/Core/M5StickC/20191118__StickC_A04_3110_Schematic_Rebuild_PinMap.pdf