m5stack/M5-Schematic #11: SIM800L Module As A

ianwillianb, 2019-08-18:
Is it possible to use the M5Stack integrated speaker/amplifier to output sounds coming from the SIM800L without any modifications to the board? 

The microphone is sensitive enough to pickup voices in a distance of 1 meter? 

Does the microphone eliminates noises?

f4grx, 2019-10-18:
Hello

I am not a m5stack member but I had a look at the schematic and the answer is "probably possible", since the speaker is just connected to GPIO 25, and this pin is available on the m5bus

It remains to be seen if, in the schematic of the sim800 module, the audio output can be routed to the pin for gpio 25.

Note that the bus shows that GPIO 26 and 25 are stereo audio, but the speaker is only connected to gpio25. There is no stereo, just 2 DAC outputs.

f4grx, 2019-10-18:
According to the schematic for the sim800 module, the speaker IS really connectable to gpio25, if resistor R2 is installed:

https://github.com/m5stack/M5-Schematic/blob/master/Modules/module_sim800l_sch.pdf

However the microphone on the sim800 module is standalone.
Noise elimination characteristics are then probably ONLY related to the sim800 configuration.

ianwillianb, 2019-10-21:
@f4grx Thanks for the reply, glad that is possible.

closed 2019-10-21 by ianwillianb